dragino/Arduino-Profile-Examples #1: Stuck at rf95.waitPacketSent() in Simple Client Arduino

sebasira, 2017-08-17:
Hi!

I'm using a LG01 as server and a LoRa Bee as client. The Bee connected to an ATMega328 with 3.3v running at 8Mhz (custom board like the Arduino Pro Mini) the Simple Client Arduino get always stuck after sending, at rf95.waitPacketSent();

The ATMega328 and Bee connection  is:
SCK to SCK
MOSI to MOSI
MISO to MISO
Bee CS to D10

I'm using RadioHead Library v1.79 (the lattest right now)

I hope you can help me here, this is my frist time with LoRa modules

dragino, 2017-08-17:
You still need DIO0 to the ATMega328P D2 pin. This is for RX interrupt

sebasira, 2017-08-17:
Thank you very much! I didn't know about that. That has solved my problem!

closed 2017-08-17 by sebasira