dragino/Arduino-Profile-Examples #7: LG-01 Sensor Data is gibberish

geofbaum, 2018-04-18:
After attempting to connect to TTN, and somehow having it work once, my gateway is no longer connected and any attempt at sending data from a node is pretty much junk. Either way, I followed the wiki page, http://wiki.dragino.com/index.php?title=Connect_to_TTN and specifically looked at the 915Mhz section for my node which oddly enough doesn't seem to be the problem as I can see the correct data reaching the gateway through serial. Following through the, TTN setup, uploaded the v003 of the packet forwarder hex file and then rebooted. Followed that up by setting my radio settings for a correct frequency, 902.3 is what I used, and then adding router.us.thethings.network as my server, server port 1700, and my gateway id as the  eui id for the gateway from TTN without the eui- prefix. Still nothing, and as I mentioned it was connected once, which I verified through the API, http://noc.thethingsnetwork.org:8085/api/v2/gateways/eui-b61a3cc70cd5f0b3 , though I'm not even sure anymore how it connected that one time and now is no longer connecting. Running through the RX/TX frequencies for TTN hasn't seemed to help either so I'm at a loss now.

pkjavali, 2018-11-08:
Hi,
I am beginner on this. Did you ever figure out how to interpret this data? In my case, I am getting following sensor data. Could you please help me how to decipher this?
_0000000_ 0000 0000 0000 0000 005a 4603 feff 5817
0000010 a820 c9f2 d0ec 0200                    
_0000017_
Thank you!

geofbaum, 2018-11-13:
@pkjavali Sorry I haven't had a lot of time to go back and look at what readout looks like now. Since I've posted this I'm able to connect to TTN and my own server backend now so if I can remember to do so I'll try to check the output from the gateway now to see how things look.

haidarm, 2020-01-31:
i have the same problem just add 
void setup() {
  
  Serial.begin(115200);
  Serial.begin(115200);
  Serial.println(F("Starting"));
to ur setup method will fix the encoded message and go to tool 
[image]
and fix the encoding message and upload will fix ur issue

haidarm, 2020-02-11:
@geofbaum select the right serial port (9600 0r 115200)