dragino/Arduino-Profile-Examples #10: 2 or more fields @ thingspeak

ericmatgithub, 2019-01-09:
hi all

code works fine on my LG01.
field1 is smoothly updated on api.thingspeak

i am struggling adding 2 or more fields:

just uncommenting these two lines:
                    //dataString ="field2=";
                    //dataString += h;
doensnt make sense.
Then I loose field1. 
>> as datastring is refilled with field2 in stead of field1

furthermore 2nd line must be hh in stead of h. That's clear :-) 

anyone suggestions how to update the code in an easy / logical way?

thanks
eric

dragino, 2019-01-10:
Bug fix, new code: 
                 dataString ="field1=";
                    dataString += th;
                    dataString +=".";
                    dataString += tl;
                    dataString +="&field2=";
                    dataString += hh;
                    dataString +=".";
                    dataString += hl;

More efficient can use MQTT code, check manual.

closed 2019-01-10 by dragino

ericmatgithub, 2019-01-10:
great, this should work :-)